luni64/VisualTeensy #2: New library manager

luni64, 2018-09-26:
V0.6 which should fix the library issue

[VisualTeensy.zip](https://github.com/luni64/VisualTeensy/files/2421176/VisualTeensy.zip)

WMXZ-EU, 2018-09-27:
copied VisualTeensy.zip but VisualTeensy. exe does not start (or crashes)
(maybe wrong net library)
is in develop latest code?

WMXZ-EU, 2018-09-27:
can you re-post the mods for the LIB_OBJ for 'old library structure'
I cannot find the code anymore

luni64, 2018-09-27:
> getLibs(@"C:\Arduino\arduino-1.8.7\hardware\teensy\avr\libraries"); or similar
> replaced string by 'libPath'

Sorry, was working on the Library manager already..  I'll exchange the exe. ... and I'll wrap the app in a try catch block to get more information if it crashes :-/  

> can you re-post the mods for the LIB_OBJ for 'old library structure'
> I cannot find the code anymore

Sorry, don't understand your  request. If you need the current makefile "tail" you find it here `https://github.com/luni64/VisualTeensy/blob/develop/VisualTeensy/Embedded/makefile` 

Yes, latest code is in develop

luni64, 2018-09-27:
Just had a look but I didn't check that version in. Anyway, I didn't change much

[image]

If you remove the marked part (without the last closing bracket)  the /utility subdir is not included and you can add it manually. I don't think I changed anything else regarding the libs. 
Second line handles the src folder you can comment that out completely if you don't want it.

luni64, 2018-09-27:
Updated the zip in fist post

closed 2018-10-01 by luni64