luni64/VisualTeensy #7: Localized Get-Date error avoid linkage

Pendragon, 2018-11-25:
Hello,

As a French user, I was unable to use the generated Makefile as is. 
Get-Date in France returns a comma separated list due to localized number separator. The conversion to int fails after that, and linkage fails.

`[LD]  bin\newProject.elf
c:/progra~2/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe:--defsym:1: syntax error
collect2.exe: error: ld returned 1 exit status
`
I suggest to change the generated line 

`FLAGS_LD  := -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=$(shell powershell [int](Get-Date -UFormat +%s))`

into

`FLAGS_LD    := -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=$(shell powershell [int](Get-Date -UFormat +%s)[0])`

Please note the [0] that takes the first argument of a table.
This should work if the separator is a . or a ,

Regards

luni64, 2018-11-25:
Very cunning :-),
Thanks for finding this one. Do you need that fixed imediately or can you wait for the next update (probably tomorrow)

luni64, 2018-11-27:
Fixed in the current version

closed 2018-11-28 by luni64