luni64/VisualTeensy #10: Tools > USB Type > Serial + MIDI

elemenofi, 2019-03-17:
Hej!

Thanks for this. I am quite invested in Visual Studio Code as I use it as my main IDE for work. I am a web developer by trade but I do quite a bit of DIY MIDI projects at home. I have been trying to get a similar workflow with the Teensy to the one I have with the Leonardo when using VSC.

Your Visual Teensy has helped me a lot but I am still struggling to understand how to make CTRL+SHIFT+B to upload the code to Teensy but also set it to USB Type Serial + MIDI, right now I have to use the Arduino IDE to do that and upload the code and its a bit frustrating as I would prefer to have everything in VSC.

Is there anything I can do to help you support this use case? I love everything about this solution otherwise, thanks so much for putting the time!

luni64, 2019-03-17:
Can you explain exactly what does not work? The generated makefile would help you can cut and paste it from the preview window. 

[image]

elemenofi, 2019-03-17:
Duh!!! I completely ignored that dropdown, didnt see it!

Sorry about that man, didnt want to waste your time.

Now it all works except im getting an error at the end! 

`[LD]  .vsteensy/build/caribe.elf
User code built and linked to libraries

[HEX] .vsteensy/build/caribe.hex
   text    data     bss     dec     hex filename
  18144    2404    4876   25424    6350 .vsteensy/build/caribe.elf
Sucessfully built project

[LST] caribe.lst
Listfile generated

Teensy Loader could not find the file caribe
quitmake: *** [makefile:148: upload] Error 1
The terminal process terminated with exit code: 1`

elemenofi, 2019-03-17:
I also fixed that by closing the teensy loader and letting VisualTeensy do the whole thing for me.

It all works now! thanks so much :)

closed 2019-03-17 by elemenofi

luni64, 2019-03-18:
Good to hear that it is working now.
I strongly suggest to use [TyTools ](https://github.com/Koromix/tytools) for uploading. It can handle more than one board and has a nice serial monitor integrated

elemenofi, 2019-03-18:
thanks for the suggestion, i will check it out. but im such a noob when it comes to c/c++ and how makefiles and so on work that I'd rather keep it as it is while it works :)

luni64, 2019-03-18:
This is just a matter of installing TyTools (you get the installer from the link I posted)  and enter the TyTools directory in VisualTeensy settings. 
In VSC you then choose "Upload (TyTools)" instead of "Upload(Teensy uploader)". Thats it. You'll love it :-)

elemenofi, 2019-03-18:
I got it working but I dont understand where the serial monitor is :S

elemenofi, 2019-03-18:
aha I found the little tab in the tytools app, but cant I have the monitor in my terminal in VSC?