luni64/VisualTeensy #11: New Project name and path?

quezadaminter, 2019-04-08:
Hi, I have been up and down the VT page trying to find out how to assign a new project's name and path with VisualTeensy v0.9.1. The documentation seems to match v0.1 which shows entries for these values which seem to be missing in v0.9.1. I'm sure there is a simple answer, would you please point out where I am not looking? :)

Thanks.

[image]

luni64, 2019-04-08:
Sorry, the documentation is definitely not up to date. 
The project name is simply the name of the project folder. If you want to change the path/name just close VT, move the folder to any convenient place and re open it with VT.

Please note: the "Save As" dialog needs some improvements (avoid using it to avoid frustration) The easiest way to generate a new project is to first generate an empty project folder with the explorer,  open this folder with VT, choose settings and save.

quezadaminter, 2019-04-10:
Thanks! I guess I did not explore the Open Project menu option since it seemed to me the project would have to exist :). Thanks for the great work!

closed 2019-04-10 by quezadaminter