luni64/VisualTeensy #12: Only Teensy 3.x and LC are displayed in the board menu

weirdocollector, 2019-04-14:
Hi!

As described in the subject: can't select my Teensy 2.00++ as the board to be used when generating the make file.

I'm using release 0.91 on WIn10 with the Arduino IDE and the Teensyduino addon installed.

I've also tried to run VisualTeensy as administrator to no avail.

Thanks for abny help,

luni64, 2019-04-14:
Hi, 
sorry, VisualTeensy only generates projects for T-LC, T3.1 - T3.6. Give a T-LC a try. It is cheap and so much faster than a T2.0.

weirdocollector, 2019-04-15:
Thanks!

In your YouTube video I saw that the drop down menu showed also the 2.0 boards and I assumed that I was doing something wrong.

Case closed.

Thanks again.

oomek, 2019-08-31:
Could you explain why Teensy 2.0 support has been dropped? I can't use anything else than Teensy 2.0 in my project and I would like to try your app.

luni64, 2019-09-01:
In fact it wasn't dropped. It was never implemented since I didn't think that there is interest for the T2. In the video the filtering of the list of supported boards was simply wrong. 

Anyway, supporting Teensy 2 might be not too much effort. If you have time and are willing to do some experiments I can have a look.

oomek, 2019-09-01:
Sounds great. Thank you.

luni64, 2019-09-01:
Ok, 
The problem is that T2 uses a completely different toolchain. I did a first test and committed that to the develoment branch. Can you give it a try (I assume you can compile a visual studio c# project? if not, let me know) 

The generated makefile should work in principle you currently need to edit the path to the compiler manually. Just replace "amr" with "avr" in the GCC_BASE line (line 25). See picture below. 
Compiles for me, but I don't have a T2 to test if it works. 

[image]

oomek, 2019-09-01:
I'm gonna try it in the evening. Thanks

luni64, 2019-09-09:
Still interested in that? Otherwise I suggest to close the issue...

oomek, 2019-09-10:
I tried it and I got
`GCC_BASE         := vtCore.CheckedPath`
is that normal?

oomek, 2019-09-10:
[image]

luni64, 2019-09-10:
Can you try if it works without copy core files?

luni64, 2019-09-10:
Ok, the problem is that the Teensy2 not only needs the files in the core folder but also the files in the usb_*** folders below the core folder. They are not copied to the project folder which generates the problem. Fixing that will be some work and has low priority so please don't expect a quick solution. 

If you deselect the "embed in project" checkbox VisualTeensy uses the core files from the arduino folder and everything works.  (don't forget to change the GCC_BASE path to avr instead of amr as shown [here](https://github.com/luni64/VisualTeensy/issues/12#issuecomment-526912045)

oomek, 2019-09-10:
That's fine. I can keep using the buggy PlatformIO for the time being. No need to rush it.

nico-paris, 2019-12-16:
Hi there! First off, thank you so much for VisualTeensy! It greatly improves my coding efficiency.

Now, I've been able to compile the development branch from source and get VisualTeensy to create Teensy2.0 projects that compile without issue however when I try to upload with teensy.exe or TyCommander I get the following errors:

[image]
[image]

The generated tasks.json appears to be the same as my Teensy3 projects so maybe the issue is that the arguments to VisualTeensy-dev/make.exe need to be different for Teensy2.0?

luni64, 2019-12-16:
It looks like the spaces betweeen the arguments to TyCommander vanished? Can you post the generated makefile?

nico-paris, 2019-12-16:
Here it is (had to add the .txt to get github to let me upload)
[makefile.txt](https://github.com/luni64/VisualTeensy/files/3969190/makefile.txt)

luni64, 2019-12-16:
Yes, the spaces in the tycmd arguments are missing in your makefile. Here how it should look like (line 92)
[image]

Can you please add the spaces and try if the upload with tyCommander works then? Did you change anything when you compiled the dev-branch?

nico-paris, 2019-12-16:
Awesome. I added the spaces but now I'm having trouble getting it to find the hex file.

TARGET_HEX should be the right path, but when I try to upload with the following settings, I get an error code from tyCommander (ignore the Problems, they're just Intellesense being dumb)
[image]

Now when I change $(TARGET_HEX) to point directly at the generated file (.vsteensy/build/unitConversionDMX.hex) it succeeds
[image]

TARGET_HEX should expand to .vsteensy/build/unitConversionDMX.hex so I'm not sure where it's failing

I didn't make any changes as far as I'm aware when I compiled the branch.

luni64, 2019-12-16:
You can print out the value of variables with info. E.g. place the following before your rules:

[image]

This will print out the value of TARGET_HEX. 
>I didn't make any changes as far as I'm aware when I compiled the branch.

I can try to reproduce but this will take some time (next weekend) This whole Teensy2 thing is very experimental, there might be some bugs in the code indeed.

luni64, 2019-12-16:
For debugging you can also remove the @ before the upload commands in the rule section.  This will print the actual command  (@ suppresses this)

nico-paris, 2019-12-16:
Thanks for the help!

TARGET_HEX was being set correctly, however it gets set *after* UPL_TYCMD is assigned thus it was passing an empty string to TyCommanderC.

I moved the Binaries section to immediately follow the Folders and File section and upload task for both teensy.exe and TyCommander now works!

nico-paris, 2019-12-16:
Summary of things I did to go from building -dev and getting everything to work:
- removed "-test " from the UPL_PJRC line in Binaries
- added spaces between arguments on the UPL_TYCMD line in Binaries
- moved the Binaries subsection to follow the Folders and Files subsection
- (unrelated to your code) Teensy 2.0 support is still in beta in TyCommander so I had to add a new environment variable TYTOOLS_EXPERIMENTAL_BOARDS to my PATH. Setting it to anything should be sufficient; I set it to 1

luni64, 2019-12-17:
Thanks for the list of required changes. Very useful indeed. I'll see if I can fix that as soon as I have some time.

closed 2020-02-12 by luni64