luni64/VisualTeensy #13: GCC complains about library names

vectorcrumb, 2019-08-06:
I can't tell if this is patched in 0.9.5 (my computer complains when trying to run VisualTeensy.exe), but with 0.9.1 there's a bug when using libraries with spaces in their names. For example, when trying to use the PCD8544 and GFX libraries from Adafruit, the `LIBS_LOCAL` variable in the makefile is generated as: 
```
LIBS_LOCAL       := Adafruit GFX Library Adafruit PCD8544 Nokia 5110 LCD library
```
However, later, when iterating over `LIBS_LOCAL` to compile these files, each separate word is taken as a different library. I managed to fix this by placing underscores in the name, like so:
```
LIBS_LOCAL       := Adafruit_GFX_Library Adafruit_PCD8544_Nokia_5110_LCD_library
```
However, I don't know C#, so I can't tell if this was fixed for the current beta release.

luni64, 2019-08-06:
Thanks for reporting that. I didn't expect libraries to have spaces in the name (silly me...) I'll have a look how to fix this later today. 

Would also be interested in why the 0.9.5 doesn't run on your PC. Can you post some error messages?

luni64, 2019-08-06:
Ok, found the reason for that, will take a couple of days to fix it. Stay tuned

luni64, 2019-08-10:
Fixed in the latest version 0.9.6.0 on the development branch.

closed 2019-08-12 by luni64