luni64/VisualTeensy #15: Feature Request: Linking libraries from any local folder

logdog, 2020-01-15:
Currently libraries can be linked from the `Teensyduino Libraries`, the `Arduino Repository`, and the `Installed, Shared Libraries` locations. I would like to be able to choose any folder from a drop down menu and select libraries from there, and then generate the appropriate makefile and copy to the proper `.vsteensy/build/lib/` folder. 

I think this could easily be accomplished by changing whatever value modifies the `LIBS_SHARED_BASE` (line 20 in makefile) location on the C# side of things and letting VisualTeensy run its course. Included below is where the drop down menu would be included, along with displaying the path to the selected folder below.

[image]

luni64, 2020-01-16:
I don't quite understand if you want to have  additional (user selectable) source folders for
a) "shared" libraries or 
b) libraries copied to the project folder 

a) would be a major change since this would require a redesign of the generated makefile. b) could be implemented quickly.

logdog, 2020-01-21:
I have multiple git repos, each with their own Teensy libraries. I wanted to be able to use VisualTeensy with these different libraries while they are still in development. 

Option c) sounds like it is the best work-around. Ideally, VisualTeensy would allow me to link multiple libraries from multiple locations with option a). This would require a key-value pair system like

```python
{
 'C:\Users\...\Documents\Code' : ['MotorController', 'LED_Helper', ...],
 'C:\Users\...\Downloads': ['Accelerometer', ...],
 'E:\gitstuff\code' : ['CustomStepperMotor', ...]
}
```
where the keys were the paths and the values were a list of libraries used from that particular path (as shown above). This would be a major redesign of the makefile (I'm not even sure if this is possible with makefiles) but this would convenient for including libraries that are still in development.

luni64, 2020-01-21:
> I have multiple git repos, each with their own Teensy libraries. I wanted to be able to use VisualTeensy with these different libraries while they are still in development.

In this case I usually use the following structure
```
project
    src
        main.cpp    
    lib
        libUnderDeveloment
        otherLib
```

and clone the 'libUnderDevelopment' directly in the lib folder. Thus you can conveniently commit/push library changes. 

To set up VisualTeensy you edit .vsteensy/vsteensy.json and add the name of the library folder to the projectLibraries list. Here an example where I did this with the TeensyTimerTool library

[image]

Save everything, open VisualTeensy (the lib should appear in the library tab. If you now generate the project VisualTeensy generates the makefile including the lib.  
You can do this for as many libraries you like. For me this is the most convenient way to develop libraries since F12 and all the other lookup functions work seamlessly and I can easily push changes to github. 

Hope I got your use case right...

logdog, 2020-01-23:
Yes! Thank you for that thorough example. That is exactly how I plan on developing libraries. For clarification, your `TeensyTimerTool` would be structured in the project as such:
```
project
    src
        main.cpp
    lib
        TeensyTimerTool
```
where I copied the `TeensyTimerTool` from its original location using a mv command or equivalent?

luni64, 2020-01-23:
> For clarification, your TeensyTimerTool would be structured in the project as such:
Yes

>where I copied the TeensyTimerTool from its original location using a mv command or equivalent?

You can copy of course, I personally clone the library from GitHub directly in this place so pushing back changes, switching branches etc is just a mouse click away. 

NOTE: Avoid having the same lib in the project folder (as above) AND in your standard Arduino library folder. VisualTeensy will confuse them in this case.  This is a bug in the makefile which I never bothered to fix...  If you need to have the same library in both places for your workflow the easiest is to rename the "development copy" of your lib. Or nag me long enough to fix the bug :-)

logdog, 2020-01-23:
Thanks so much for your help!

closed 2020-01-23 by logdog