luni64/VisualTeensy #17: Is it compatible with vsciot-vscode.vscode-arduino extension?

ackvf, 2020-02-11:
I am total noob so forgive me this question.

Is this project compatible with [Arduino extension for VSCode](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino)?

Or is the extension totally unnecessary? (I am not even sure yet what it offers.)

---

I have just finished tutorial 2 on pjrc https://www.pjrc.com/teensy/tutorial2.html , and I immediately realized how limited the Arduino IDE is. So, upon searching and going through forums I ended up here https://forum.pjrc.com/threads/48878-How-to-add-Teensy-board-to-VS-Code-with-Arduino-plugin which asks for a Teensy integration to the aforementioned Arduino extension.

It seems that your solution is completely unrelated and at first I was consufed whether I could use your solution at all and it took me a while to realize that the `.cpp` you use in your examples are just renamed `.ino` files after having went through this thread https://forum.pjrc.com/threads/53604-Compiling-Teensy-Sketches-with-VisualCode-(Win10)?p=187220&viewfull=1#post187220

Would you perhaps consider mentioning some of the stuff you mention in the thread here in github?

luni64, 2020-02-11:
>Is this project compatible with Arduino extension for VSCode?
Or is the extension totally unnecessary? (I am not even sure yet what it offers.)

I tried the Arduino extension some years ago but it didn't really work. VisualTeensy is not at all related to this. In fact VisualTeensy is not a vsCode extension, it just generates all necessary files to build a Teensy project with vsCode. 

>Would you perhaps consider mentioning some of the stuff you mention in the thread here in github?

Can you be more specific? What exactly do you want to know?

ackvf, 2020-02-11:
For example, I have only realized I can use your solution to generate my project from this comment [here](https://forum.pjrc.com/threads/53604-Compiling-Teensy-Sketches-with-VisualCode-(Win10)?p=187983&viewfull=1#post187983) where you mention to rename `.ino` to `.cpp`. Until then, I was very confused and thought that it's incompatible, because the structure in Arduino IDE `.ino` files is little different from what is in VSCode with your generated files.

_Yes, I didn't know the language used is C++ :), I have only just started._

PS: I have already finished tutorial 1 and 2 on pjrc with your solution and it's great, so thanks!! Right now I am configuring `.editorconfig` and git versioning. Would you mind answering a quick OT question? Does it make sense to commit `.vsteensy/build/core` folder or not? I really just want to version both source and resulting `.hex` file (so that I don't need to rebuild), but I don't need overly complicated `.gitignore`.

luni64, 2020-02-11:
>For example, I have only realized I can use your solution to generate my project from this comment here where you mention to rename .ino to .cpp. Until then, I was very confused and thought that it's incompatible, because the structure in Arduino IDE .ino files is little different from what is in VSCode with your generated files.

Got you. When I did VisualTeensy I decided not to use the .ino extension since the Arduino IDE is doing things to .ino's like automatic forward declaration etc., which I find awful and  didn't want to replicate. So VisualTeensy generates  normal c++ projects with normal .cpp an .h files. You are right, that should be documented somewhere. Generally the documentation for this project is a quick hack, if I find some time I'll improve it. 

>... Does it make sense to commit .vsteensy/build/core folder or not? I really just want to version both source and resulting .hex file (so that I don't need to rebuild), 

No,  it absolutely makes no sense to commit binaries. Rebuilding a normal project after pulling is a matter of about 10s. Hex is fine, it allows users to just download the firmware without a need to build the project. 

>but I don't need overly complicated .gitignore.

I usually simply copy a gitignore from somewhere and add the missing parts. Here a working file
[gitignore.zip](https://github.com/luni64/VisualTeensy/files/4187723/gitignore.zip) Don't worry about a complicated .gitignore, they are parsed in milliseconds...

Have a look at GitHub Desktop, it makes working with git much easier SourceTree is also working nicely but is a bit more complicated

ackvf, 2020-02-11:
Excellent, thank you for your insight!

closed 2020-02-12 by luni64