luni64/VisualTeensy #18: Add support for custom generated files (static assets)

ackvf, 2020-02-12:
Now that I am going through the Teensy tutorial, it is apparent that I will always have a need to copy certain files to each new project, such as `.editorconfig` and `.gitignore`.

I can think of two solutions:

**A) Easier** - Introduce a folder within VisualTeensy called `assets` which will hold custom static assets that will always be copied to new project. Perhaps if a folder structure is introduced, the files would be merged to the respective folders in the new project.

**B)** Add an inapp editor which could reuse existing elements to set the _file content_ and configure _path_
[image]

_PS: When I thought about the option B, I was under the assumption that the Generated Files text area is editable._

luni64, 2020-02-12:
Thats a **very** good idea. I find myself copying .clang-format to the projects. I'll implement something like that in the next version

luni64, 2020-02-12:
Added a wish list to the issues (see #19)

closed 2020-02-13 by ackvf