luni64/VisualTeensy #20: How to set up project for multiple boards?

ackvf, 2020-02-23:
I believe I have already read it somewhere but I can't seem to find it again, or perhaps I am mistaking it for something else...

How can I easily configure the project to compile to both, Teensy LC and Teensy 3.2?

luni64, 2020-02-23:
Currently VisualTeensy does not directly support this but it is already on the wishlist and backlog https://github.com/luni64/VisualTeensy/projects/1

Here an old forum post where I show how to do this semi manually
https://forum.pjrc.com/threads/53604-Compiling-Teensy-Sketches-with-VisualCode-(Win10)?p=187990&viewfull=1#post187990

ackvf, 2020-02-24:
Oh yes!, I knew I saw it somewhere, thanks for finding it.