luni64/VisualTeensy #22: Add general option and/or per library setting to save shared libs to the project

luni64, 2020-03-08:
> I could find it interesting to create a complete standalone development system (not only core, but also all the libraries) so there are no multiple dependency issues.

@WMXZ-EU  
https://forum.pjrc.com/threads/57245-VisualTeensy-upgraded-to-support-Teensy-4-projects?p=232320&viewfull=1#post232320

That is already working for all but shared libraries.

cbaugher, 2022-02-21:
Bump for this feature.  I need to compile from a local-to-project library instead of the normal shared library because of a naming conflict between two shared libraries.  Being able to disable the use of shared libraries completely, and only compile from local-to-project libraries would be useful as well, for tighter source control.

C|

luni64, 2022-02-21:
As a workaround you can copy the library to the lib folder (don't add it in VisualTeensy).  Then start VisualTeensy and open the project. It will detect the manually copied library and add it to the relevant jsons.  Save the project and you should be ready to go.

cbaugher, 2022-02-21:
> As a workaround you can copy the library to the lib folder (don't add it in VisualTeensy). Then start VisualTeensy and open the project. It will detect the manually copied library and add it to the relevant jsons. Save the project and you should be ready to go.

I tried that but VT is detecting both libraries (local and shared) and I still get class redefinition errors on compile.  Do I need to delete some files manually before running VT (or something)?

C|

luni64, 2022-02-21:
Can you post the generated makefil

cbaugher, 2022-02-21:
> Can you post the generated makefile?

Sure, included below.  Thanks for the quick assistance BTW!

`#******************************************************************************
# Generated by VisualTeensy V1.4.0 on 2/21/2022 at 1:52 PM
#
# Board              Teensy 4.1
# USB Type           Serial
# CPU Speed          600 MHz
# Optimize           Faster
# Keyboard Layout    US English
#
# https://github.com/luni64/VisualTeensy
#******************************************************************************
SHELL            := cmd.exe
export SHELL

TARGET_NAME      := MotorInterfaceBoard
BOARD_ID         := TEENSY41

MCU              := imxrt1062

LIBS_SHARED_BASE := C:\Users\Chris\Documents\Arduino\libraries
LIBS_SHARED      := Adafruit_MCP23017_Arduino_Library TMCStepper Adafruit_INA260_Library ArduinoJson Adafruit_BusIO MillisTimer 

LIBS_LOCAL_BASE  := lib
LIBS_LOCAL       := Adafruit_BusIO-1.9.9 Adafruit_INA260_Library-1.5.0 Adafruit_MCP23017_Arduino_Library-2.0.2 MillisTimer-1.0.0 OneWire SPI TMCStepper-0.7.3 Wire 

CORE_BASE        := C:\PROGRA~2\Arduino\hardware\teensy\avr\cores\teensy4
GCC_BASE         := C:\PROGRA~2\Arduino\hardware\tools\arm\bin
UPL_PJRC_B       := C:\PROGRA~2\Arduino\hardware\tools

#******************************************************************************
# Flags and Defines
#******************************************************************************

FLAGS_CPU   := -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16
FLAGS_OPT   := -O2
FLAGS_COM   := -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD
FLAGS_LSP   := 

FLAGS_CPP   := -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing
FLAGS_C     := 
FLAGS_S     := -x assembler-with-cpp
FLAGS_LD    := -Wl,--print-memory-usage,--gc-sections,--relax -T$(CORE_BASE)/imxrt1062_t41.ld

LIBS        := -larm_cortexM7lfsp_math -lm -lstdc++

DEFINES     := -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO_TEENSY41 -DARDUINO=10813
DEFINES     += -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH

CPP_FLAGS   := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_CPP)
C_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_C)
S_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_S)
LD_FLAGS    := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_LSP) $(FLAGS_LD)
AR_FLAGS    := rcs
NM_FLAGS    := --numeric-sort --defined-only --demangle --print-size

#******************************************************************************
# Colors
#******************************************************************************
COL_CORE    := [38;2;187;206;251m
COL_LIB     := [38;2;206;244;253m
COL_SRC     := [38;2;100;149;237m
COL_LINK    := [38;2;255;255;202m
COL_ERR     := [38;2;255;159;159m
COL_OK      := [38;2;179;255;179m
COL_RESET   := [0m

[... 179 lines cut ...]
	@$(NM) $(NM_FLAGS) "$<" > "$@"

%.hex: %.elf
	@echo $(COL_LINK)[HEX] $@
	@$(OBJCOPY) -O ihex -R.eeprom "$<" "$@"

# Cleaning --------------------------------------------------------------------
cleanUser:
	@echo $(COL_LINK)Cleaning user binaries...$(COL_RESET)
	@if exist $(USR_BIN) rd /s/q "$(USR_BIN)"
	@if exist "$(TARGET_LST)" del $(subst /,\,$(TARGET_LST))

cleanCore:
	@echo $(COL_LINK)Cleaning core binaries...$(COL_RESET)
	@if exist $(CORE_BIN) rd /s/q "$(CORE_BIN)"
	@if exist $(CORE_LIB) del  $(subst /,\,$(CORE_LIB))

cleanLib:
	@echo $(COL_LINK)Cleaning user library binaries...$(COL_RESET)
	@if exist $(LIB_BIN) rd /s/q "$(LIB_BIN)"

# compiler generated dependency info ------------------------------------------
-include $(CORE_OBJ:.o=.d)
-include $(USR_OBJ:.o=.d)
-include $(LIB_OBJ:.o=.d)`

cbaugher, 2022-02-23:
So I did finally get it to work.  I had to delete all the VT generated files and run it again with all the libraries copied to the local lib folder.  If you don't delete the old files first it seems to get confused.

Anyway I think I'm good for now, thanks!

C|