luni64/VisualTeensy #25: Unable to start

csmith-Deltek, 2020-05-25:
Clean PC, new install of VSCode, installed c/c++ extension.  downloaded VisualTeensy v0.9.7.3 started VisualTeensy.exe.  Windows asks for Arduino folder or leave empty.  I leave empty and click Start VisualTeensy, get the error attached.  (Win10 Pro x64 v1909)

[image]

Please advise.

luni64, 2020-05-25:
Can you try to enter the Arduino folder?

Vission01, 2020-05-25:
Had not installed arduino at all.  I found that after installing Arduino it now works even without choosing a folder.

This e-mail and any attachments are intended only for the named recipient(s) and may contain information that is legally privileged, confidential, or exempt from disclosure under applicable law. This message may be logged for archival purposes, may be reviewed by parties at Deltek<http://www.deltek.com/> other than those named in the message header, and may not necessarily constitute an official representation of Deltek<http://www.deltek.com/>. If you have received this message in error, or are not the named recipient(s), you may not retain copy or use this e-mail or any attachment for any purpose or disclose all or any part of the contents to any other person. Any such dissemination, distribution or copying of this e-mail or its attachments is strictly prohibited. Please immediately notify the sender and permanently delete this e-mail and any attachment from your computer.

You are responsible for complying with all applicable data protection, import, re-import, export, and re-export control laws, including any applicable license requirements, and country-specific sanctions programs. Without limiting the foregoing, you are solely responsible for compliance related to the information you send, via email, to Deltek, its employees, or agents. Please contact legalcompliance@Deltek.com with any questions or concerns.

luni64, 2020-05-26:
Great, hope it works stable now for you

closed 2020-05-26 by luni64