luni64/VisualTeensy #26: VisualTeensy does not start

Huebenthal, 2020-05-28:
After entering the Arduino folder (or even leave it empty) I just get a fault report that some object link does not point on a object instance (see attached screen shot).
[image]
OS: Windows 10 64bit

luni64, 2020-05-28:
This is very strange, is this on first startup when it asks you to enter an Arduino Folder?

Huebenthal, 2020-05-28:
It appears when I press "Start VisualTeensy" on the first screen where the path to Arduino is set up. Brand new installation.

luni64, 2020-05-29:
Sorry for my late answer, I was traveling yesterday...

It is always difficult to debug things which only appear on fresh installations...  I suspect that some of the required files will be generated by Arduino after a first use. So my suggestions would be to try to compile and upload a simple blink.ino with the Arduino IDE first and try if you can start VisualTeensy afterwards.

Huebenthal, 2020-06-01:
No change. I re-installed Arduino and the Teensy extension but no change either. I'll stay with the Teensy development in the Arduino IDE.

luni64, 2020-06-02:
Using it for daily work on various PCs without issues. Works for others as well.  
Best wishes

closed 2020-06-02 by luni64