luni64/VisualTeensy #29: *.ino mode compatibility to portable installations

luni64, 2020-10-30:
see: https://forum.pjrc.com/threads/53604-Compiling-Teensy-Sketches-with-VisualCode-(Win10)?p=257609&viewfull=1#post257609

luni64, 2020-10-30:
Can you please try the attached version: 
[Uploading VisualTeensy.zip…]()

luni64, 2020-10-30:
[VisualTeensy.zip](https://github.com/luni64/VisualTeensy/files/5465110/VisualTeensy.zip)

lorenzofattori, 2020-10-30:
@luni64 this solves the problem

`Opening Teensy Loader...
Using library Ethernet at version 2.0.0 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\Ethernet 
Using library MIDI at version 5.0.2 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\MIDI 
Using library SPI at version 1.0 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\SPI
Using library SD at version 1.2.2 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\SD
Using library ArduinoJson at version 5.13.5 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\portable\sketchbook\libraries\ArduinoJson
Using library Artnet in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\Artnet (legacy)
Using library EEPROM at version 2.0 in folder: C:\Users\Lorenzo\Desktop\Arduinos\Teensy\arduino-1.8.13 - TeensyDuino 1.53\arduino-1.8.13\hardware\teensy\avr\libraries\EEPROM
"C:\\Users\\Lorenzo\\Desktop\\Arduinos\\Teensy\\arduino-1.8.13 - TeensyDuino 1.53\\arduino-1.8.13\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-size" -A "C:\\Users\\Lorenzo\\Desktop\\test_visualteensy\\Mm2q_V2.27_MidiBeatClock\\Mm2q_V2.27_MidiBeatClock\\.vsTeensy\\build/Mm2q_V2.27_MidiBeatClock.ino.elf"
Sketch uses 51956 bytes (4%) of program storage space. Maximum is 1048576 bytes.
Global variables use 8000 bytes (3%) of dynamic memory, leaving 254144 bytes for local variables. Maximum is 262144 bytes.`

luni64, 2020-10-30:
Very good, 

here the release: https://github.com/luni64/VisualTeensy/releases/tag/v0.9.9.1