luni64/VisualTeensy #31: include lib folder into include search path

Fuzzlix, 2020-11-09:
First: Nice work!
It works better for me than any other solution i tried.

Your makefile already generates the include-parameters for gcc but vscode editor cant find the included files yet.

wish:
* c_cpp_properties.json: include `"lib/**"` into `includePath.`
* create lib folder, even it it empty.
* optional: place a README in the lib folder, describing how including work automatically.

luni64, 2020-11-09:
I'm a bit confused. vsTeensy is supposed (and does) add the correct include paths to c_cpp_properties automatically if you use the library tab. Can it be that you copied the libraries manually? In that case just reopen the project in vsTeensy, it will detect the new libs and adjust all files accordingly. You should see the manually copied libs in the library tab after this. 

> create lib folder, even it it empty.
optional: place a README in the lib folder, describing how including work automatically.

Good idea, I can do that in the next release.

luni64, 2020-11-09:
BTW: sometimes (seldom) vsCode doesn't detect changes in c_cpp_properties when it runs while the changes are done which can be quite confusing.  Restarting vsCode always fixes it.

Fuzzlix, 2020-11-11:
The reason i asked, is:
I want to use some non Arduino Modules ( [lua](lua.org), [FreeRTOS](FreeRTOS.org), ...) on a striped down Arduino source skeleton.
Your Makefile is the quick and easy way to get things compiling. I like the way, how you auto detect the lib/ sub folders. 

BTW: I am using a Teensy4.1.

luni64, 2020-11-13:
Looks like this is working for you now? Can we close this?

closed 2021-04-18 by luni64