luni64/VisualTeensy #33: Shared Library Path

dabryant, 2020-12-10:
Is there a way to set the shared library path?  

The default path is:
`LIBS_SHARED_BASE := C:\Users\*username*\Documents\Arduino\libraries`

To make the makefile more portable between different systems I have been using:  
`LIBS_SHARED_BASE := $(USERPROFILE)\Documents\Arduino\libraries`

I am doing this manually right now, but it gets overwritten every time I update the anything in VisualTeensy.

Thanks for a great tool!

luni64, 2020-12-10:
Arduino users can set the shared library path in the arduino settings file (preferences.txt -> sketchbookfolder + "/libraries"). So, the path might be different for different users.  If I'd set it to  `C:\Users\*username*\Documents\Arduino\libraries` it would not work for anybody with a different setting in preferences.txt. 

Workaround:
You can however set it using the expert settings / additional entries to makefile:

[image]

Of course, this only changes the makefile and not `c_cpp_properties.json` which is needed to set up intellisense. -> Your code will compile but intellisense probably won't find your libraries. 

Let me know if this fixes your issue.

dabryant, 2020-12-10:
Makes sense about the path not being the default one for everyone.  I think the best way for me going forward is going to be to copy those shared libraries to the project folder.  That makes this problem go away and has the added benefit of the libraries being under source control.

Thanks for the help, and thanks again for a great tool!

closed 2020-12-10 by dabryant

luni64, 2020-12-10:
> Makes sense about the path not being the default one for everyone. I think the best way for me going forward is going to be to copy those shared libraries to the project folder. That makes this problem go away and has the added benefit of the libraries being under source control.

Good idea indeed. If you just copy / clone them to the lib folder,  make sure to re-read and save the project once with vsTeensy. It will detect the new libs and adjust the makefile and the *.jsons automatically.