luni64/VisualTeensy #35: Error following quick start guide

makeller1, 2021-01-02:
Hello, I followed the quick start guide but I am getting the following error after running the Build task:

"
> Executing task: F:/CppLearning/VisualTeensy_v0_9_9_2/make.exe all -j -Otarget <

USER [CPP] main.cpp
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\teensy\avr\cores\teensy4: No such file or directory
make: *** [makefile:230: .vsteensy/build/src/main.o] Error 1
make: *** Waiting for unfinished jobs....
CORE [AR] .vsteensy/build/core.a
Teensy core built successfully

The terminal process terminated with exit code: 1
"

I checked in the makefile code and c_cpp_properties.json and the path for teensy4 in the code is "K:/Program Files (x86)/Arduino/hardware/teensy/avr/cores/teensy4". I'm not sure why the path is getting shortened (little experience with json). 
If it's not immediately obvious what's happening here, can someone suggest a way to debug make.exe?

Thanks!

luni64, 2021-01-02:
The problem seems to be the space in your Arduino installation folder which is strange since I thought I'd catch all of those. Can you post the contents of  vsteensy.json and c_cpp_properties.json?

makeller1, 2021-01-02:
Ah I see. Here's c_cpp_properties.json:
`{
  "configurations": [
    {
      "name": "VisualTeensy",
      "includePath": [
        "src/**",
        "K:/Program Files (x86)/Arduino/hardware/teensy/avr/cores/teensy4"
      ],
      "forcedInclude": [],
      "defines": [
        "__IMXRT1062__",
        "TEENSYDUINO=153",
        "ARDUINO_TEENSY40",
        "F_CPU=600000000",
        "USB_SERIAL",
        "LAYOUT_US_ENGLISH",
        "ARDUINO"
      ],
      "compilerPath": "K:/Program Files (x86)/Arduino/hardware/tools/arm/bin/arm-none-eabi-gcc.exe",
      "intelliSenseMode": "gcc-arm",
      "cppStandard": "gnu++14"
    }
  ],
  "version": 4
}`

and vsteensy.json:
`{
  "version": "1",
  "target": "vsCode",
  "buildSystem": "makefile",
  "debugSupport": "cortex_debug",
  "configurations": [
    {
      "name": "default",
      "guid": "1e23c86a-072d-4af1-ad5b-703c43992936",
      "setupType": "quick",
      "coreBase": "K:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr",
      "compilerBase": "K:\\Program Files (x86)\\Arduino\\hardware\\tools\\arm",
      "makefileExtension": null,
      "copyCore": false,
      "sharedLibraries": [],
      "projectLibraries": [],
      "board": {
        "name": "Teensy 4.0",
        "options": {
          "USB Type": "Serial",
          "CPU Speed": "600 MHz",
          "Optimize": "Faster",
          "Keyboard Layout": "US English"
        }
      }
    }
  ]
}`

luni64, 2021-01-02:
Sorry, forgot the makefile, can you post this as well?

makeller1, 2021-01-02:
Sure. 
```

#******************************************************************************
# Generated by VisualTeensy (https://github.com/luni64/VisualTeensy)
#
# Board              Teensy 4.0
# USB Type           Serial
# CPU Speed          600 MHz
# Optimize           Faster
# Keyboard Layout    US English
#
# 1/1/2021 9:23 PM
#******************************************************************************
SHELL            := cmd.exe
export SHELL

TARGET_NAME      := TestTeensy2
BOARD_ID         := TEENSY40

MCU              := imxrt1062

LIBS_SHARED_BASE := C:\Users\Mathew\Documents\Arduino\libraries
LIBS_SHARED      := 

LIBS_LOCAL_BASE  := lib
LIBS_LOCAL       := 

CORE_BASE        := K:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4
GCC_BASE         := K:\Program Files (x86)\Arduino\hardware\tools\arm\bin
UPL_PJRC_B       := K:\Program Files (x86)\Arduino\hardware\tools
UPL_TYCMD_B      := K:\Program Files (x86)\Arduino\hardware\tools

#******************************************************************************
# Flags and Defines
#******************************************************************************
FLAGS_CPU   := -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16
FLAGS_OPT   := -O2
FLAGS_COM   := -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD
FLAGS_LSP   := 

FLAGS_CPP   := -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing
FLAGS_C     := 
FLAGS_S     := -x assembler-with-cpp
FLAGS_LD    := -Wl,--print-memory-usage,--gc-sections,--relax,--defsym=__rtc_localtime=$(shell powershell [int](Get-Date -UFormat +%s)[0]) -T$(CORE_BASE)/imxrt1062.ld

LIBS        := -larm_cortexM7lfsp_math -lm -lstdc++

DEFINES     := -D__IMXRT1062__ -DTEENSYDUINO=153 -DARDUINO_TEENSY40 -DARDUINO=10807
DEFINES     += -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH

CPP_FLAGS   := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_CPP)
C_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_C)
S_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_S)
LD_FLAGS    := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_LSP) $(FLAGS_LD)
AR_FLAGS    := rcs
NM_FLAGS    := --numeric-sort --defined-only --demangle --print-size

#******************************************************************************
# Colors
#******************************************************************************
COL_CORE    := [38;2;187;206;251m
COL_LIB     := [38;2;206;244;253m
COL_SRC     := [38;2;100;149;237m
COL_LINK    := [38;2;255;255;202m
COL_ERR     := [38;2;255;159;159m
COL_OK      := [38;2;179;255;179m
COL_RESET   := [0m

#******************************************************************************
[... 179 lines cut ...]

%.hex: %.elf
	@echo $(COL_LINK)[HEX] $@
	@$(OBJCOPY) -O ihex -R.eeprom "$<" "$@"

# Cleaning --------------------------------------------------------------------
cleanUser:
	@echo $(COL_LINK)Cleaning user binaries...$(COL_RESET)
	@if exist $(USR_BIN) rd /s/q "$(USR_BIN)"
	@if exist "$(TARGET_LST)" del $(subst /,\,$(TARGET_LST))

cleanCore:
	@echo $(COL_LINK)Cleaning core binaries...$(COL_RESET)
	@if exist $(CORE_BIN) rd /s/q "$(CORE_BIN)"
	@if exist $(CORE_LIB) del  $(subst /,\,$(CORE_LIB))

cleanLib:
	@echo $(COL_LINK)Cleaning user library binaries...$(COL_RESET)
	@if exist $(LIB_BIN) rd /s/q "$(LIB_BIN)"

# compiler generated dependency info ------------------------------------------
-include $(CORE_OBJ:.o=.d)
-include $(USR_OBJ:.o=.d)
-include $(LIB_OBJ:.o=.d)
```

luni64, 2021-01-02:
Weird. 

```
CORE_BASE        := K:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4
GCC_BASE         := K:\Program Files (x86)\Arduino\hardware\tools\arm\bin
UPL_PJRC_B       := K:\Program Files (x86)\Arduino\hardware\tools
UPL_TYCMD_B      := K:\Program Files (x86)\Arduino\hardware\tools
```
should be shortened by a OS call to something like:
```
CORE_BASE        := C:\toolchain\Teensyduino\current_fork\cores\teensy4
GCC_BASE         := C:\PROGRA~1\Arduino\ARDUIN~1.12\hardware\tools\arm\bin
UPL_PJRC_B       := C:\PROGRA~1\Arduino\ARDUIN~1.12\hardware\tools
UPL_TYCMD_B      := C:\toolchain\TyTools
UPL_JLINK_B      := C:\PROGRA~2\SEGGER\JLINK_~1
```

Is `K` a normal Windows folder or maybe something unusual? Win10 and current Version of vsTeensy right?

makeller1, 2021-01-02:
I've never heard otherwise if it's not.. It is Win10 and the current vsTeensy. I will try changing the arduino location to a path with no spaces and report back.

luni64, 2021-01-02:
> I've never heard otherwise if it's not.. It is Win10 and the current vsTeensy. I will try changing the arduino location to a path with no spaces and report back.

Nevertheless, I'd like to fix this. Can you do me a favor and try the following?

Open a command prompt and do a dir /x in the root folder. You should see something like:
[image]

makeller1, 2021-01-02:
Here you go:
[image]

luni64, 2021-01-02:
Interesting, looks like Windows doesn't want to assign 8.3 filenames for this drive. Google finds this:

https://knowledge.kofax.com/Capture/Kofax_Capture/Scan/Enable_8dot3_File_Name_Creation#:~:text=To%20enable%208dot3%20File%20Name%20Creation%20from%20the%20command%20prompt%3A&text=Execute%20the%20following%20command%3A%20fsutil,Reboot%20the%20PC. 

Maybe it was disabled?

sutaburosu, 2021-01-02:
That shows how to enable it for all partitions.  That may not be desirable.  [It is possible to enable this per-partition](https://michlstechblog.info/blog/windows-enable-generation-of-8-3-names/).  Win10 only enables this on C: by default.

luni64, 2021-01-02:
Perfect, thanks for the link

closed 2021-01-14 by luni64