luni64/VisualTeensy #36: ARDUINO define not given a version

Bankst, 2021-01-27:
the ARDUINO define in the c_cpp_properties.json is not given a version, and this causes the Intellisense to fail as the core libraries check this value to be greater than 100.

changing `"ARDUINO"` to `"ARDUINO=151"` fixes the Intellisense.

luni64, 2021-01-27:
Unfortunately this number can't be read out anywhere I'm aware of. I can however set it to a fixed value like you did. Probably better than not setting it at all...

sutaburosu, 2021-01-27:
Perhaps you could grep it out of [$arduino_path/hardware/teensy/avr/cores/teensy4/Makefile](https://github.com/PaulStoffregen/cores/blob/master/teensy4/Makefile#L69-L70) at startup.

luni64, 2021-01-27:
Thanks for the find. Thinking of it I'd rather not add a dependency to an Arduino Installation. Currently, VisualTeensy can be used without even installing Arduino which I'd like to keep. 

Anyway, I added changed c_cpp_properties.json to have "ARDUINO=1813" now and uploaded to the master branch. In case a higher number will be needed later (unlikely) I can update this manually. Are you able to compile the app yourself or do you need the binaries?

sutaburosu, 2021-01-27:
That file is installed by TeensyDuino,  which I think is a hard requirement of VisualTeensy.  I dunno.  I thought Arduino was required too.  :)

Regardless, I think you're missing a digit in your define.  It should be `10813`. There's at least one line in TeensyDuino with `#if ARDUINO >= 10600`, so this will probably make a difference.

@Bankst do you need a binary?

Bankst, 2021-01-27:
> That file is installed by TeensyDuino,  which I think is a hard requirement of VisualTeensy.  I dunno.  I thought Arduino was required too.  :)
>
> Regardless, I think you're missing a digit in your [...]

You are correct, it should be `10813`. I didn't catch that as my project only has `#if ARDUINO >= 100` checks. At this point I've already heavily tweaked the makefile for my purposes, just thought future users would benefit.

luni64, 2021-01-27:
Actually, you only need to clone the Teensyduino repository and install GCC to be up and running. 

I took the number from your post above, anyway, changed it to 18013 and upload to master

sutaburosu, 2021-01-27:
That's good to know.  Thanks.

I'm not sure it matters much,  but that should be `10813` not `18013`. :)

luni64, 2021-01-27:
Not my day :-) I'll fix it later

closed 2021-01-29 by luni64