luni64/VisualTeensy #37: Project structure with multiple programs

beyon, 2021-01-28:
Thank you for this project. I've been using it to generate the base when doing larger teensy projects and then manually editing it for my needs.

When working on larger projects what I've found I need is a shared code base with many programs/main.cpp.

This is roughly the project structure I'm currently using:

```
/project-name
  /programs
    /some-program
      /.vscode
      /.vsteensy
      /src
      makefile
    /some-other-program
    /tests
      /test-program
      /subsystem-test-program
  /shared
    /libs <- teensy libs
    /src <- project source code shared between programs
```
This works ok after having set it up for one program and then copying that folder for new programs and applying minor manual edits. I then work with the project folder open in one vscode window and then opening a specific program folder in a new window to build/run that program.

Do you have a good way to use visual teensy to structure projects in this way or is this something you would consider adding that would fit within the scope of the project?

luni64, 2021-01-28:
You will probably understand that everyone and their grandmother have different ways to structure a project. Thus it doesn't seam feasible to to implement such things in a general purpose software. 

Anyway, I'd like to understand your structure a little bit more. Which of those folders contain teensy code besides the obvious some-program folder?

beyon, 2021-01-29:
I understand that there is no ONE way to structure your projects but possibly the general purpose software can be a bit more general. The structure I currently use and mentioned is most likely ideal even for me, however the thing I've found to repeatedly need/want is many applications/programs/main.cpp using the same shared teensy libs and project specific code. 

Maybe it would have been a better idea to start with a discussion about project structure on the PJRC forums than as an issue here?

Not sure where you draw the line of what to count as teensy code, all code in the project intended to run on teensy even if parts might work on other platforms too.

```
programs/some-program
programs/some-other-program
programs/tests/test-program
programs/tests/subsystem-test-program
```
Contain the kind of project VisualTeensy generates but without lib folder (containing copies of Teensy libs) which is instead moved to `shared/lib` (seems I wrote libs by mistake in previous post). The makefile, c_cpp_properties.json, vsteensy.json in those dirs have been manually edited to make it work.

Project specific code shared between programs is in `shared/src`.

Does that explain it?

So while I think it would be a good addition to VisualTeensy to support some kind of more involved/alternate project generation I completely understand if it doesn't fit the project.

If some way that can benefit more people than me can be found I might be able to try help implementing it.

luni64, 2021-01-29:
OK, I think I understand. Thanks for the info. 

**Shared libs:** 
You probably know that the shared libraries currently are read from sketchbookfolder/libraries. It probably would help the location of this folder would be settable per project. You could then set it to your project specific folder. 
That should be doable without too much effort. I'll think of it. 

**Shared code:** 
Couldn't you simply treat your shared code as library? I.e. just copy the shared code it to a folder in the shared library's folder (together with the point above  this would be within your project). You should then be able to select it in the library tab of vsTeensy. 

**more than one subproject in a main project (your some-program folders)**
This should work out of the box just use VisualTeensy to generate a project in each of the folders. BTW: there are extensions to define different color schemes per project which is quite helpful if you work on projects in parallel (e.g. client/server architectures), In case you need to compile all of those subprojects with one command you can always manually place a makefile in the root folder /project-name which simply calls the makefiles in your subprojects. 

I'm currently working (slowly) on a library mode which allows to set up library projects. I.e., one main project with some examples (which are self-contained projects. Since this is similar to your needs I might generate some spinoffs supporting your scenario.