luni64/VisualTeensy #45: Suggested content for Quick Start Guide

benpeart, 2021-04-20:
I'm new to Teensy so am starting fresh with nothing installed. I was successful in building and flashing my images using the Arduino IDE but when I switched to VSCode+VisualTeensy I ran into some problems I think could be easily avoided.

I was able to generate a new project using VisualTeensy and successfully build it but when I got to the step that said "To upload to the TeensyBoard you can use one of the uploaders shown in the Terminal->Run Build Task menu." I didn't have any uploaders in my build task menu.

I eventually found out where to install them from but it would have been great to have that step listed in the "Installation" section at the beginning of the quick start guide with a link to the latest release like you have done for VisualTeensy (https://github.com/Koromix/tytools/releases) .

Note, I installed the 32 bit version the first time and got an error every time I tried to run it from the build task:

```
> Executing task: & 'C:/Program Files/VisualTeensy/make.exe' uploadTy -j -Otarget <

process_begin: CreateProcess(NULL, C:\PROGRA~2\TyTools /tyCommanderC upload .vsteensy/build/HelloBlinkingWorld.hex --autostart --wait --multi, ...) failed.make (e=5): Access is denied.
make: *** [makefile:167: uploadTy] Error 5
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command & 'C:/Program Files/VisualTeensy/make.exe' uploadTy -j -Otarget" terminated with exit code: 1.
```

I eventually figured out to uninstall the 32 bit version and install the 64 bit version and then it worked fine so you may want to mention this in the quick start guide as well.

luni64, 2021-04-21:
Thanks for the input. I Added information about installing additional uploaders in the quick start guide. 

> ... I didn't have any uploaders in my build task menu.

This is strange. You should at least have the stock PJRC uploader when you told VisualTeensy the Arduino location at its first start up.  Anyway, glad it works now.

benpeart, 2021-04-22:
Honestly, my memory is hazy - I'm not sure if the teensy uploader was there and just didn't work (see #46) or if it was missing. I'll hope/assume the former. :)

The additional information in the installing section looks good.

closed 2021-04-22 by benpeart