luni64/VisualTeensy #46: Unable to upload image using teensy.exe build task

benpeart, 2021-04-20:
When I use the "Upload (teensy.exe) build task, I get the following error:

```
> Executing task: & 'C:/Program Files/VisualTeensy/make.exe' upload -j -Otarget <

Teensy Loader could not find the file HelloBlinkingWorld
quitmake: *** [makefile:164: upload] Error 1
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command & 'C:/Program Files/VisualTeensy/make.exe' upload -j -Otarget" terminated with exit code: 1.
```

I have been unable to figure out why it is unable to find my binary (HelloBlinkingWorld in this case). The tycommander build task uploads fine in this same project. I've created several different projects trying to figure out why it is failing but without success.

[image]

luni64, 2021-04-21:
Interesting, I can have a closer look if you post the following files: 

.vsteensy/vsteensy.json
.vscode/c_cpp_properties.json
and the generated makefile

luni64, 2021-04-21:
I'm wondering why it uses the powershell instead of the usual cmd.exe to execute make.  Did you apply a special Windows setting for this?

dontfeedthenerd, 2021-04-21:
I have the same problem both with TYcommander and the PJRC.
I have attached my files

[c_cpp_properties.zip](https://github.com/luni64/VisualTeensy/files/6354320/c_cpp_properties.zip)

luni64, 2021-04-22:
Unfortunately I can't reproduce this here. Your files look perfectly normal.  Can you please open the makefile and remove the @ in lines 164 and 165:

[image]

You then should see the actual command issued when it tries to upload the file. Check if the path and filename correspond to the location of your hex image.  You can also copy the line into a new vscode terminal and  execute it manually. Screenshots of those experiments would be helpful.

dontfeedthenerd, 2021-04-22:
[image]
Looks like the .hex isn't being appended to the -file?

dontfeedthenerd, 2021-04-22:
Nope that's not it... hmmm. I think it may need the full path to-file?

luni64, 2021-04-22:
Just to make sure: Your project is called teensyproj and you find the file `teensyproj.hex` in  the `vsteensy/build` folder, right? 
Did you try to manually call the command from a vscode terminal? Does it work then?

dontfeedthenerd, 2021-04-22:
That is correct I can find a teensyproj.hex
As a matter of fact I've been using VisualTeensy to build then manually
loading using the teensyloader.

luni64, 2021-04-22:
>As a matter of fact I've been using VisualTeensy to build then manually
yes but did you try exactly the command which is shown in your screenshot? 

This works here since years without issue. Very weird. What is the output of a call to TyCommander?

dontfeedthenerd, 2021-04-22:
[image]
This is the output to TyCommander let me try running it manually

dontfeedthenerd, 2021-04-22:
[image]

This is the output while running manually

dontfeedthenerd, 2021-04-22:
did it in a command prompt running as admin same issue and here's proof the .hex exists:
[image]

dontfeedthenerd, 2021-04-22:
Thank you for your help!

This is the result of that:
[image]

It just hangs I have to ctrl+C out of it

dontfeedthenerd, 2021-04-22:
oh wait here we go:
[image]

luni64, 2021-04-22:
Ok, now I can reproduce it. You obviously removed the "" around the command otherwise powershell reports an errror right? If I do the same I get the file not found issue. Looks like this only happens if vscode uses the powershell terminal. Can you switch to a cmd terminal and try if it works then? 

open the terminal combobox and select "Select Default Profile" 
[image]

Select "Command Prompt" (instead of the currently selected PowerShell) 

[image]

Close all open termials:
[image]

and try to upload....

dontfeedthenerd, 2021-04-22:
No Joy :(

[image]

dontfeedthenerd, 2021-04-22:
[image]
tried running the command manually

luni64, 2021-04-22:
does it even find the file if you do
type .vsteensy\build\teensyproj  in the vscode terminal?

dontfeedthenerd, 2021-04-22:
it does
[image]

luni64, 2021-04-22:
Sorry, need to sleep over it. Maybe something changed in the call parameters of the teensy uploader. Can you check what your Arduino installation spits out to upload the hex (need to switch to verbose output to see it).

dontfeedthenerd, 2021-04-22:
Yeah take your time, I don't mind using teensy to upload

benpeart, 2021-04-22:
Switching to "Command Prompt" as the default terminal window got the teensy uploader working for me. Would be nice if it worked in powershell as well but at least I have a good workaround. Thank you!

closed 2021-08-23 by luni64

nmccoy, 2023-05-11:
For anyone else searching this in the future: I had the same error message but a different root cause. Turns out I just needed to close the teensy.exe loader window opened by the Arduino app.