luni64/VisualTeensy #49: Prebuilt Visual Teensy 1.0.0.0 fails to start when deserializing additionalFiles

AmadeusW, 2021-05-26:
After downloading and unzipping [Visual Teensy 1.0.0.0 from GitHub's tags page](https://github.com/luni64/VisualTeensy/releases/tag/v1.0.0.0), program crashes on startup with error in deserializing `additionalFiles`
[image]

This does not repro when I build locally.

I thought there is a difference in `.config` files, but neither downloaded nor built `VisualTeensy.exe.config` contains the `additionalFiles` entry.

luni64, 2021-05-26:
Looks like there is a bug when upgrading an existing user config file.  Can you try to delete your user config file? It lives here in  AppData\Local\lunOptics.  (here: C:\Users\lutz\AppData\Local\lunOptics). Best to delete the whole contents of the folder.

AmadeusW, 2021-05-28:
This worked. I removed (renamed for backup) the lunOptics directory and was able to start from scratch and without issues.

luni64, 2021-05-28:
Thanks a lot for testing this. I had a brief look at the code yesterday and it looks like the setup file has problems with initializing the List<string> I use for the "additional files". I'll check if the problem is on my side and look for a fix.

luni64, 2021-08-23:
fixed in v1.2.0

closed 2021-08-23 by luni64