luni64/VisualTeensy #50: --defsym=__rtc_localtime=$(shell powershell [int](Get-Date -UFormat +%s)[0]) not accepted

sschiesser, 2021-05-28:
When using the prebuilt visualTeensy 1.0.0.0 example (Win10, 1909), I get the following error message:

`[LD]  .vsteensy/build/visualTeensyTest.elf 
c:/progra~2/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe:--defsym:1: syntax error
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:236: .vsteensy/build/visualTeensyTest.elf] Error 1
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'C:/Program Files (x86)/VisualTeensy/make.exe' all -j -Otarget" terminated with exit code: 1.`

It looks like ld.exe doesn't like the syntax of the --defsym flag. When replacing the whole $(...) part with 1 (result given by the powershell), it doesn't complain at all.

Is that a Windows or a visualTeensy issue? Thanks!

luni64, 2021-05-28:
This is interesting, I assume you are sure that you didn't accidentally changed the makefile?  Can you remove the @ before the linker command in the makefile and recompile so that I can see the actual link command? 

[image]

sschiesser, 2021-05-28:
To be sure, I generated a new project with the following settings:
***@***.******@***.***D75409.AD2B64A0]

Then it generated according to the following settings:
***@***.***D75409.AD2B64A0]

…and I got the error message:
***@***.***D75409.AD2B64A0]

By removing the @ before the linker command, it looks that the shell process doesn’t want to start:
***@***.***D75409.AD2B64A0]

Does it help you? (me not really ;-) ).

Thanks for your quick help.

Best,
Sebastien

—
—
Zürcher Hochschule der Künste
Zurich University of the Arts
—
Sébastien Schiesser
+41 43 446 55 06
—
Research Associate
Institute for Computer Music and Sound Technology
www.icst.net
—
Studio Manager
Immersive Arts Space
www.immersive-arts.ch
—
Toni-Areal, Pfingstweidstrasse 96
P.O. Box, CH-8031 Zurich
—
www.zhdk.ch

[image]<https://user-images.githubusercontent.com/12611497/119995869-5a0e8000-bfce-11eb-9c35-bcec7672a3c1.png>

luni64, 2021-05-29:
Looks like you answered by eMail which seems to have thrown off the attached screenshots.  Would it be possible to repost your answer directly in GitHub?

I can only guess that something in your configuration is different, I got a few complaints about weird things happening with the scripting recently. I assume that something in vsCode, Powershell or Windows changed but I wasn't able to reproduce it so far.  I'm highly interested in fixing this but I need to be able to reproduce the effect first. 

Some questions which might help:
- Do you have any special settings in vsCode? Maybe a non standard shell?  
- Can you post your vsCode user / workspace  setting files (`xxx\AppData\Roaming\Code\User\setup.json`  and ( if exist) `yourProject\.vscode\settings.json`)

sschiesser, 2021-05-29:
Updated my previous comment in github to see pictures... now they appear as links (!)

My setup.json look pretty simple:
```
{
    "editor.formatOnSave": true,
    "[cpp]": {
        "editor.defaultFormatter": "ms-vscode.cpptools"
    },
    "editor.tabSize": 2
}
```

My project has a `tasks.json` and a `c_cpp_properties.json` but no `settings.json`

Makefile:
```
#******************************************************************************
# Generated by VisualTeensy (https://github.com/luni64/VisualTeensy)
#
# Board              Teensy 3.6
# USB Type           Serial
# CPU Speed          180 MHz
# Optimize           Faster
# Keyboard Layout    US English
#
# 29 May 2021 11:07
#******************************************************************************
SHELL            := cmd.exe
export SHELL

TARGET_NAME      := newProject
BOARD_ID         := TEENSY36

MCU              := mk66fx1m0

LIBS_SHARED_BASE := C:\PROGRA~2\Arduino\hardware\teensy\avr\LIBRAR~1
LIBS_SHARED      := 

LIBS_LOCAL_BASE  := lib
LIBS_LOCAL       := 

CORE_BASE        := C:\PROGRA~2\Arduino\hardware\teensy\avr\cores\teensy3
GCC_BASE         := C:\PROGRA~2\Arduino\hardware\tools\arm\bin
UPL_PJRC_B       := C:\PROGRA~2\Arduino\hardware\tools
UPL_TYCMD_B      := C:\PROGRA~2\Arduino\hardware\tools

#******************************************************************************
# Flags and Defines
#******************************************************************************
FLAGS_CPU   := -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant
FLAGS_OPT   := -O2
FLAGS_COM   := -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD
FLAGS_LSP   := 

FLAGS_CPP   := -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti
FLAGS_C     := 
FLAGS_S     := -x assembler-with-cpp
FLAGS_LD    := -Wl,--print-memory-usage,--gc-sections,--relax,--defsym=__rtc_localtime=$(shell powershell [int](Get-Date -UFormat +%s)[0]) -T$(CORE_BASE)/mk66fx1m0.ld

LIBS        := -larm_cortexM4lf_math -lm -lstdc++

DEFINES     := -D__MK66FX1M0__ -DTEENSYDUINO=154 -DARDUINO_TEENSY36 -DARDUINO=10813
DEFINES     += -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH

CPP_FLAGS   := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_CPP)
C_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_C)
S_FLAGS     := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_S)
LD_FLAGS    := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_LSP) $(FLAGS_LD)
AR_FLAGS    := rcs
[... 193 lines cut ...]

%.hex: %.elf
	@echo $(COL_LINK)[HEX] $@
	@$(OBJCOPY) -O ihex -R.eeprom "$<" "$@"

# Cleaning --------------------------------------------------------------------
cleanUser:
	@echo $(COL_LINK)Cleaning user binaries...$(COL_RESET)
	@if exist $(USR_BIN) rd /s/q "$(USR_BIN)"
	@if exist "$(TARGET_LST)" del $(subst /,\,$(TARGET_LST))

cleanCore:
	@echo $(COL_LINK)Cleaning core binaries...$(COL_RESET)
	@if exist $(CORE_BIN) rd /s/q "$(CORE_BIN)"
	@if exist $(CORE_LIB) del  $(subst /,\,$(CORE_LIB))

cleanLib:
	@echo $(COL_LINK)Cleaning user library binaries...$(COL_RESET)
	@if exist $(LIB_BIN) rd /s/q "$(LIB_BIN)"

# compiler generated dependency info ------------------------------------------
-include $(CORE_OBJ:.o=.d)
-include $(USR_OBJ:.o=.d)
-include $(LIB_OBJ:.o=.d)
```

luni64, 2021-05-29:
That all looks OK. I saw that you have the shared libraries folder inside the Arduino folder which is strange (usually it is at ...Documents/arduino/libraries) but this is certainly not the problem. 

To isolate the problematic call to `get-date` I slightly rewrote the makefile. Maybe this gives more info about the underlying problem.  Can you replace  the FLAGS_LD .... line (around line 45) by the following 3 lines: 

```
TIME_SYM    := $(shell powershell [int](Get-Date -UFormat %s))
$(info TIME_SYM: $(TIME_SYM))
FLAGS_LD    := -Wl,--print-memory-usage,--gc-sections,--relax,--defsym=__rtc_localtime=$(TIME_SYM) -T$(CORE_BASE)/mk66fx1m0.ld
```

It also looks like your powershell is quite old. In your error message you get: 
```C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'C:/Program Files... ``` 

If I provoke a linker error I get:  

```C:\Program Files\PowerShell\7\pwsh.exe -Command... ```

Maybe updating to a current version helps

sschiesser, 2021-05-31:
Hi

First the rewroten Makefile doesn't look to behave on a different way than before. The terminal gives:
[image]

Second I noticed that Windows had a powershell version 5.1 (x86) located somewhere in `/AppData/Roaming/...`. So I updated it to v7.1 (located now in `/Program Files/PowerShell` and uninstalled v5.1.
Unfortunately, the result is still the same:
[image]

luni64, 2021-05-31:
Thanks a lot that helped. The issue might be related to the setting of the decimal char.  Can you type the following in a powershell terminal (e.g. the standard terminal in vscode) 
```
Get-Date -UFormat %s 
```
On a PC with e,g, German language setting this gives e.g. 1622464559,36875 (decimal symbol comma) on a PC with englisch settings I get 1622464559.36875 (dot). However, the cast to int doesn't work with the comma. 
So, if I type this 
```
[int](Get-Date -UFormat %s)
```
command into the terminal, I get this error:
```
Cannot convert value "1622464685,06447" to type "System.Int32". Error: "Input string was not in a correct format."
At line:1 char:1
+ [int](Get-Date -UFormat %s)
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvalidCastFromStringToInteger
```
On a PC with a dot as decimal symbol I get the correctly casted result.

Can you check if 
```
Get-Date -UFormat %s
```
outputs the date with dot or comma as decimal separator?

BTW: Sorry to bother you with debugging my code....

sschiesser, 2021-06-01:
Weird: I get an int value
[image]

But my settings on this topic are customized and therefore probably leading to some mistake:

- Keyboard language: Swiss german
- Input language: English (Switzerland)
- Decimal symbol: (dot)
- Digit grouping symbol: (apostrophe)

...don't know if that helps...

BTW: no problem with helping debugging your code. Thank you to allow me working with teensy in a real IDE ;-)

luni64, 2021-06-01:
>Weird: I get an int value

Actually it returns a string and it looks like the standard formatting  for this string changed between PS versions.  Anyway, the following should return an integer (number of seconds since 1970)  regardless of the PS version and/or language settings. I tested it with swiss settings and all available combinations of decimal and grouping symbols. 
```ps
[int][double]::Parse((Get-Date -UFormat %s))
```
I'll try to compile a new  vsTeensy version for testing later today.

luni64, 2021-06-04:
Let me know if this works for you

luni64, 2021-08-23:
fixed in v1.2.0

PR #51 (2021-06-04): Fix bug with getting current date/time from powershell if custom decimal symbol is used
fixes #50
closed 2021-08-23 by luni64