luni64/VisualTeensy #67: Clone core from github into project should not bring in .git & .github folders

mechatroniks-git, 2021-11-25:
I have to delete the .git and .github folders after cloning, so github desktop will commit my stuff. Otherwise, I get the error shown below. 

I also delete the .gitignore. Perhaps keywords.txt and readme.md should stay?

[image]

luni64, 2021-11-25:
Can you give me a more complete description on how I can reproduce the issue?

mechatroniks-git, 2021-11-25:
Here are my steps, thanks for looking at this.

1. I create a blank project in v1.4.0 Visualteensy
2. I select "clone core from Github into project"
3. Save project
4. Project opens in VSCode
5. Then I go into Github Desktop to publish the project, it gives me the above error.
6. By deleting the git folders inside the Cores directory, Github Desktop allows me to publish the repo

I compared other projects I had (using visualteensy 1.2.0 & "clone local core into project" and noticed these extra files/folders.

luni64, 2021-11-26:
Thanks,  I'll have a look why Github Desktop stumbles over it. 
Anyway, the '.git  folder' is the whole point of cloning the library. Deleting it renders the clone to a simply snapshot copy. Looks like you want to select Copy instead of clone? 

[image]

luni64, 2021-11-26:
The following applies if you really want to **clone** the core files into your project. If you want to **copy** them do as mentioned above. 

----

I did a few experiments and can reproduce the issue you have with GitHub Desktop. If you clone the core libraries into a project which you want to upload to gitHub (or any other git server) later, you need to tell your project that it should treat the cloned cores folder as a git submodule. Unfortunately the git library I use can't currently do this automatically so there is a small manual step required: 

1) Clone the core files into the project with VisualTeensy. This will first git-init your project and then add the Teensy core as git repository to it. 

[image]

2) In VSC, open a terminal and type `git submodule add https://github.com/paulstoffregen/cores ./cores` this will 'register' the cloned core library as submodule to the main repo

[image]

3) In Github Desktop choose File | Add local Repository and browse to your project folder. Don't forget to commit

[image]

Then, use the publish repository button to generate the repo in GitHub.  In Github you will see that the cores folder is empty. Actually it simply links to the location it was cloned from originally. I.e. github.com/paulstoffregen/cores . When you clone your repo from Github it automatically grabs the core files from there
[image]

Hope that helps

closed 2022-02-01 by luni64