luni64/VisualTeensy #70: Error uploading with tyCommander

finncat62, 2022-02-24:
Hi Luni, this is the error that I posted on your youtube. Sorry about that!

I'm using a teensy 3.2

In this video, I show myself opening visual teensy from scratch, all of the filepaths, and the error message. If you want to see something else, I can easily take another video or more photos. Thank you so much for the help!
https://www.youtube.com/watch?v=xohOADM4qxg

Visual teensy load page:
[image]
Visual Teensy Settings
[image]

Visual studio code environment with blink program. Error code:
 "Successfully built project
[SYM] .vsteensy/build/test1.sym
process_begin: CreateProcess(NULL, C:\Dropbox\ALLSAV~1\robotics\1ARDUI~1\VISUAL~1\unzipped\tytools/tyCommanderC upload .vsteensy/build/test1.hex --autostart --wait --multi, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [makefile:169: uploadTy] Error 2
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'C:/Dropbox/all saves/robotics/1Arduinorecent/visualTeensy/unzipped/make.exe' uploadTy -j -Otarget" terminated with exit code: 1.
"
[image]

luni64, 2022-02-24:
I noticed the following things: 

You are using an ancient version of powershell, you have spaces in your path to tyCommander and your folder seems to be some Dropbox folder. The combination might generate the issue. Can you please check the following: 

1) Install tyCommander at a normal (not dropbox etc) folder without spaces in the path (e.g. c:\toolchain\tyCommander) and check if that fixes the issue
2) your video shows that vscode prompts you to upgrade your powershell installation. Please do so and try again (might require a reboot). Current versions should be able to handle spaces. 

3) There also was a similar issue here #35, I doubt that your issue has the same root cause, but if 1) and 2) don't help you can check the linked method in the last posts of this issue as well.

finncat62, 2022-03-06:
https://www.youtube.com/watch?v=jLQEkHZx2jI

Hi, sorry for taking a while. I updated powershell, changed tytools to a shorter path outside of dropbox, and I don't know what do with the error from #35.

Error code:
Sucessfully built project

[SYM] .vsteensy/build/newProject(2).sym
process_begin: CreateProcess(NULL, C:\OLD_AL~1\robotics\Arduino\tytools/tyCommanderC upload .vsteensy/build/newProject(2).hex --autostart --wait --multi, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [makefile:169: uploadTy] Error 2
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command & 'C:/Dropbox/all saves/robotics/1Arduinorecent/visualTeensy/unzipped/make.exe' uploadTy -j -Otarget" terminated with exit code: 1.

 When I opened the make file in vsCode, it was very strange:
[image]
[image]

although in the other area of makefile, it looked like this.  ¯\_(ツ)_/¯ . After the make file I have C_pp_properties.json and vsTeensy.json pasted. Sorry for the weird formatting!

#******************************************************************************
# Generated by VisualTeensy V1.5.0 on 3/6/2022 at 10:28 AM
#
# Board              Teensy 3.2 / 3.1
# USB Type           Serial
# CPU Speed          96 MHz (overclock)
# Optimize           Faster
# Keyboard Layout    US English
#
# https://github.com/luni64/VisualTeensy
#******************************************************************************
SHELL            := cmd.exe
export SHELL

TARGET_NAME      := newProject(2)
BOARD_ID         := TEENSY32

MCU              := mk20dx256

LIBS_SHARED_BASE := C:\Users\Sorin\OneDrive\Documents\Arduino\libraries
LIBS_SHARED      := 

LIBS_LOCAL_BASE  := lib
LIBS_LOCAL       := 

CORE_BASE        := C:\PROGRA~2\Arduino\hardware\teensy\avr\cores\teensy3
GCC_BASE         := C:\PROGRA~2\Arduino\hardware\tools\arm\bin
UPL_PJRC_B       := C:\PROGRA~2\Arduino\hardware\tools
UPL_TYCMD_B      := C:\OLD_AL~1\robotics\Arduino\tytools

#******************************************************************************
# Flags and Defines
#******************************************************************************
TIME_SYM    := $(shell powershell [int][double]::Parse((Get-Date -UFormat %s)))

FLAGS_CPU   := -mthumb -mcpu=cortex-m4 -fsingle-precision-constant
FLAGS_OPT   := -O2
FLAGS_COM   := -g -Wall -ffunction-sections -fdata-sections -nostdlib -mno-unaligned-access -MMD
FLAGS_LSP   := 

FLAGS_CPP   := -fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti
FLAGS_C     := 
FLAGS_S     := -x assembler-with-cpp
FLAGS_LD    := -Wl,--print-memory-usage,--gc-sections,--relax,--defsym=__rtc_localtime=$(TIME_SYM) -T$(CORE_BASE)/mk20dx256.ld

LIBS        := -larm_cortexM4l_math -lm -lstdc++

DEFINES     := -D__MK20DX256__ -DTEENSYDUINO=154 -DARDUINO_TEENSY32 -DARDUINO=10813
DEFINES     += -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH

CPP_FLAGS   := $(FLAGS_CPU) $(FLAGS_OPT) $(FLAGS_COM) $(DEFINES) $(FLAGS_CPP)
[... 276 lines cut ...]
      }
    },
    {
      "name": "Testdummy",
      "guid": "740fc1d4-c0ce-4f28-ae11-c9a40c24eb29",
      "setupType": "quick",
      "coreBase": "C:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr",
      "compilerBase": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\arm",
      "makefileExtension": null,
      "coreStrategy": "copy",
      "stdLib": "newlib",
      "sharedLibraries": [],
      "projectLibraries": [],
      "board": {
        "name": "Teensy 4.1",
        "options": {
          "USB Type": "Serial",
          "CPU Speed": "600 MHz",
          "Optimize": "Faster",
          "Keyboard Layout": "US English"
        }
      }
    }
  ]
}

luni64, 2022-03-07:
Looks like TyCommander is still installed in a folder containing spaces (spaces in paths are NEVER a good idea). Recently there was an issue with vsCode regarding paths with spaces.  Which version of vsCode are you using? The current one (v1.65) fixed this issue. Maybe this is also the reason for your strange observation. 

It also might be a good idea not to use dropbox or other special folders. Just install everything (tycommander, visualTeensy etc)  in a plain old local path without spaces or other strange stuff. E.g. in c:\toolchain or similar. (see also https://github.com/luni64/VisualTeensy/wiki/Self-Contained-Projects)

finncat62, 2022-03-13:
Next video:
Everything is in a nice short file path without spaces or weird naming anything, do you have any other Ideas as to what it could be? I reinstalled visualteensy from scratch. Maybe I did an error with that? I downloaded the zip form the github and unpacked it to a folder - was there anything else I had to do? Although that doesn't really make sense, as everything else worked. 
https://youtu.be/qKd9VD8nQj4

finncat62, 2022-03-13:
Also PS, sorry that it takes me so long to respond. I'm extremely busy and usually only have Sundays to work on the project. I'll try to get back faster!

finncat62, 2022-03-15:
Hey Luni, thank you for the help. Would it be possible to join a video call or something sometime soon? I would like to figure this out, and it should be fast once your there and able to see what exactly is going wrong. I've just been spending too much time hung up on this issue instead of the actual project, and I'm running out of time to hit my goal. 
You can contact me on instagram (@bearboy62 , profile picture is a pig), or if you give me an email(mine is sorin@jayaweera.com) I could send my phone number/ zoom link there. 

Times that I'm able to do: Any day 6-7 am Pacific Daylight Time, or any day after 6:30 pm PDT, and weekends all day basically would work. Hopefully this won't take too long, I believe it will be much faster then working it out in the error board. We can post the solution later.

luni64, 2022-03-15:
I had a look at your last video. Observations:
- VisualTeensy is working without an issue. It generates a correct makefile which runs and compiles your code.
- You **still** have TyCommander installed in a path containing spaces:  

[image]
Usually this is no problem,  VisualTeensy generates the old 8.3 file format in those cases to avoid problems.  For some reasons in your installation (vsCode, windows?) this seems to generate problems. 

[image]

The screenshot shows the correct 8.3 version of your path but vsCode complains about not finding tyCommander.exe.  For a quick check, please do as recommended a few times now and  install tyCommander in a folder without spaces and see if this fixes the issue. If so,  we can try to fix the underlying problem.  

Which version of vsCode are you using?

finncat62, 2022-03-16:
Version: 1.65.2 (user setup)
Commit: c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1
Date: 2022-03-10T14:33:55.248Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.22000

[image]

I'm so so sorry, I really thought that putting it in program files would be good because it was the same place as Arduino, and I wanted it to be organized. It was a complete brain fart moment. You've been really helpful so far, I've just been frustrated with this haha.

luni64, 2022-03-16:
Thanks for testing this. At least it removed one unknown from the equation. Can you please open the makefile and remove the @ from the call to tyCommander (around line 168). Make sure to only delete the @ (make.exe needs a tab as first char in this line)
[image]

This should display the actual call to tyCommander if you try to upload. 
I get: 
[image]

Please post a screenshot of the output you get. 

Now, copy this line to the clipboard, open a new terminal in vsCode (make sure to use a "Command Prompt" one, not one of the other terminal types vsCode offers). Copy the line into the terminal and execute it (press Enter). 
[image]

This should manually execute the command which make.exe can't execute for some reason.  Let me know the result.

luni64, 2022-03-16:
Wait, before you do this can you please try the following: 

Open the makefile and add quotation marks around `$(UPL_TYCMD_B)/tyCommanderC` in line 97. Like it is done for the Teensy Uploader in line 96. 
[image]

I remember someone having a similar issue with teensy.exe, the quotation marks fixed it (which is why they are in the makefile...)

finncat62, 2022-03-17:
Sorry, my computer crashed for a bit (overheated when I ran this or smth). I'll post this comment every few lines even if it isn't done just to save my work(I had a long report before and lost it). 
I have a photo for each step to make sure that I did it correctly, and there was no change when I used the @ or "". When using the @, I actually got an extra error. Photo dump:

luni64, 2022-03-17:
> I went to the page and clicked to download the zip

Nooo, this only contains the sources not the executable :-/ That of course explains that you can not start TyCommander, you simply didn't install it ?!? Really? Did you ever try to start it manually from this folder?

Please install the tyTools as described in Koromix's readme:
[image]

This will bring you to this page:
[image]

Download the version you need (probably win64 zip) unpack into a folder and try if it works without using vsTeensy first. If it works, use vsTeensy in the next step.

finncat62, 2022-03-18:
NOOO was it really that simple!?!?!?

[image]

Unfortunately not:(
(No changes to makefile)
[image]

"" change in makefile
[image]

luni64, 2022-03-18:
> NOOO was it really that simple!?!?!? AHHHHH YES IT WAS!!!

Great, may a ask you to try if it now works with spaces as well? I.e., do I need to add the quotation marks to the makefile or is it working without?

> 1. BIG: Code that I just worked on and saved (save button in visual studio code, not the save button in visuatlTeensy) doesn't reload when I close and re-run from visualTeensy

visualTeensy just sets up some jsons in your vscode workspace and a makefile. Addtionally it generates a default src/main.cpp if it doesn't find one. It doesn't interact with your source code at all.  Saving in visualTeensy just regenerates the jsons and the makefile and deletes the compiler output to force a new compilation. Generally you do NOT need visualTeensy after your project has been set up and can simply close it. Compiling/uploading etc is done by vsCode, not by vsTeensy. If you want to change some settings later you can open the project in vsTeensy, change the settings and save it to regenerate/update the project files. Maybe I should rename the save button to something like "Generate Project"

>    It lost changes in both main.cpp and the makefile(I copied the code from my full project into the cpp)

vsTeensy will never overwrite an existing src/main.cpp. If it does, that would be a very bad bug which a lot of people would have complained about. Can you give me detailed instructions on how to provoke the behavior you observed?
 
> When I do do the save in VisualTeensy, the code saves (sometimes, a bit inconsistent from the few times I've opened new code. I'm in bug testing mode right now to see what questions I could have for the near future) but not the makefile. Seems like its building a new make every time I run, there is no way to unselect this option: [image]
> 

Yes the makefile is generated whenever you save in vsTeensy that's the whole point of vsTeensy. You select some options in the GUI and vsTeensy is generating a corresponding makefile and the needed jsons for vsCode. 
The radio buttons you show in your screenshot are not options they just switch the preview of the generated files in the textbox below. vsTeensy will always generate the necessary files and copy them into your workspace folder. 
 
> I have libraries that are downloaded that work perfectly in the Arduino ide, but they pull up an error in vscode. !

The Arduino IDE has some automatic library search and download capability. I deliberately did not implement such a thing since IMHO this generates more problems than it solves. I wanted to have full control of the pulled in libraries. That of course means that you need to manually specify all libraries and their dependencies. Normally this is not a big deal when you start from scratch. It might be a bit tedious if you have existing code with a lot of libraries which you want to migrate to a vsTeensy project. Also see https://www.youtube.com/watch?v=8oTSou1I3IM

A quick glance at your screenshot shows that you added the required libraries. Just to make sure: You did save (generate) the project after adding the libraries? 

BTW: vsCode sometimes/rarely has issues with updating the intellisense database and will mark stuff as error which is perfectly fine. In that case closing vsCode and reopening shoud fix it. To check, you can also build the project and see if it compiles. If it compiles while vsCode shows errors it most probably is due to this intellisense database issue. 

If you zip your  original Arduino project and post it here I can check what is needed to compile it. 

> 3. Opening files often doesn't work, IE this one has code

vsTeensy is not opening files. You should point it to a folder containing your project. Again,  it does not interact with your source files, it only generates a makefile and the supporting jsons vor vsCode. 

> 4. Sometimes the library menu is there, and other times it isn't. When creating a new project it always is, but it disappears sometimes when I am opening a folder that I already have running.

>    [image]

vsTeensy only shows the lib menu if you have a makefile based project. You can switch to Arduino builder mode in the settings tab it then uses the arduino builder to compile your code (same as the Arduino IDE does). Since the builder has its own library detection the vsTeensy library tab is not needed.  I personally don't like the builder mode since it does a lot in the background which is out of control. But beginners might prefer this mode since it behaves exactly as the A-IDE does. 

> 5. This one is a basic coding question: When I have a multi board structure, each board will have it's own name.cpp. How would you recommend organizing these? Would I have all the main.cpp's in the same project, open one and run then open the other and run? Is there a different way to do this that is better? As it currently is, I don't think(maybe you can, idk your the creator) that I can open a file without the code in the next layer, and to open the code it has to be in a folder with its own name. Therefore, I don't think I can run multiple scripts and interact. OR would I just create a file and add a new to the src section. How would running that work? Basically, a long winded way of asking how to actually run master-slave protocol on here.

Usually you would use conditional compilation using #defines see e.g. here https://github.dev/luni64/cores/blob/master/teensy3/kinetis.h how you have one source code for different boards.  

> 6. Minor bug: When Libraries are open, versioning doesn't work. I click on the box for loading past versions(both in the installed and other 2 section), the box turns grey but no version numbers pop up.

I can reproduce that. It used to work, looks like something sneaked in in one of the last versions. I'll have a look

finncat62, 2022-03-20:
Libraries not loading/working properly:
All of the libraries are in visualTeensy, I saved, the project loaded without the libraries working. They are, however, added in the vsteensy.json

[image]

[image]

[image]

Also, side thing i noticed: I don't know if it was me or something else, but some libraries were added multiple times(and 2 in different versions). I deleted all of the duplicates/extras, and loaded. However, in the json there seems to be ruminants of those extras(1 example highlited).

luni64, 2022-03-22:
Not much time this morning, thus, very brief

> Also, side thing i noticed: I don't know if it was me or something else, but some libraries were added multiple times(and 2 in different versions). I deleted all of the duplicates/extras, and loaded. However, in the json there seems to be ruminants of those extras(1 example highlighted).

Never saw this. As always, I need some instructions to reproduce this if you want me to look deeper into it. 

> Libraries not loading/working properly:
All of the libraries are in visualTeensy, I saved, the project loaded without the libraries working. They are, however, added in the vsteensy.json

As mentioned above, vsTeensy does not automatically add dependent libraries. Luckily gcc is very clear in its error messages: 

[image]

It obviously complains about a missing SPI.h. So, I assume your libraries implicitly use the SPI library. -> Add this library in vsTeensy, save and try again (make sure to use the SPI library from the Teensyduino repository)

[image]

Regarding your conditional compiling question: 
I thought you want to make a source file compatible to different boards. Looks like you want to simply write two firmwares, one for a "Sender" and one for a "Receiver". In this case you'd simply generate two separate projects. Open each of them its own instance of vsCode if you want to work on both sources 'simultaneously'. tyCommander handles this perfectly. I.e, it remembers which firmware needs to go to which board so multiboard uploading "just works".

Having more than one vsCode project open can be confusing (which project am I working in? ) It helps a lot to use different colors for titlebar / side bar etc to distinguish the projects. This extension: https://marketplace.visualstudio.com/items?itemName=johnpapa.vscode-peacock is quite useful for such customizations.

finncat62, 2022-03-29:
Hi, sorry I've been busy and hadn't been able to come back. It was working, then I came back excited to get real progress, and now I'm still getting some issues when I try to actually upload:
[image]

[image]
[image]

 ^^same stuff, nothing has changed. All the filepaths are the ones that used to work, they have the correct things inside, i'm pretty sure its all good here.

opens to this:
[image]

Run tyCommander build:
[image]

[image]

Waiting for user selection
      upload@10504980-Teensy  Uploading to board '10504980-Teensy' (Teensy 3.2)
      upload@10504980-Teensy  Triggering board reboot
      upload@10504980-Teensy  Firmware: newProject(11).hex
      upload@10504980-Teensy  Flash usage: 10 kiB (3.6%)
      upload@10504980-Teensy  Uploading...
      upload@10504980-Teensy  I/O error while writing to '\\.\HID#VID_16C0&PID_0478#6&1C546ED1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}'
make: *** [makefile:169: uploadTy] Error 1
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command C:/arduinovisualteensy/visualteensy/VisualTeensy-master/VisualTeensy/make.exe uploadTy -j -Otarget" terminated with exit code: 1.

luni64, 2022-03-29:
These are error messages from TyCommander, it somehow can't write to the board. vsTeensy is obviously working, it compiles ok and tries to upload, TyCommander starts, reboots the board but can't write to it. Looks like a hardware issue. Is this permanent? Or was it a one time issue?

 Can you upload with the Teensy uploader?

finncat62, 2022-03-29:
Yeah, visualTeensy is working perfectly now, thank you so much. Could you still help me with this as a "expert" in the field? It would be great if you could, but if your busy its totally cool.
I just replaced with anew board and this popped up:
[image]

luni64, 2022-03-29:
Never seen this. Can it be that you have both the PJRC uploader and tyCommander open at the same time? The PJRC uploader starts when it registers a reboot of the board. Maybe it grabs the board from tyCommander after it has rebooted the board? 

Can also be that your hex file is somehow corrupt. Try to do a build | clean to delete it (or delete the .vsteensy/build folder manually) That will trigger a complete rebuild.

luni64, 2022-03-29:
Might also be a good idea to reboot the PC in case you have some zombie tasks lurking around. Seems you really attract problems. You should look for a job as software tester. :-)

finncat62, 2022-03-29:
"Seems you really attract problems. You should look for a job as software tester. :-)"
Saw this and laughed out loud for a good 10 seconds

I ran clean and then upload(teensy.exe)
[image]

then deleted the build folder
[image]

and rebuilt 

[image]

Restarting the computer now, will be back in a minute.

luni64, 2022-03-29:
I was able to exactly reproduce your problem when I have the PJRC uploader open while trying to upload with tyCommander...

finncat62, 2022-03-29:
cool bug ig.
I restarted, opened visual studio, tycommander opened well, 
[image]

successfully built and ran

I'll try with the other code in a second, I have to go. BRB

luni64, 2022-03-29:
Good, I added a note in the vsTeensy WIKI to make sure to close the PJRC uploader when using tyCommander (or any other uploader). https://github.com/luni64/VisualTeensy/wiki/Quick-Start-Guide

finncat62, 2022-03-29:
okay I'm back, sorry.

finncat62, 2022-03-29:
[image]

[image]

and uh, side thing - this happened:
[image]

finncat62, 2022-03-29:
(Because my long code may be hard to see what im talking about, this is the example code)
[image]

// Wire Slave Receiver
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Receives data as an I2C/TWI slave device
// Refer to the "Wire Master Writer" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

int led = LED_BUILTIN;

void setup()
{
  pinMode(led, OUTPUT);
  Wire.begin(9);                // join i2c bus with address #9
  Wire.onReceive(receiveEvent); // register event
  Serial.begin(9600);           // start serial for output
}

void loop()
{
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany)
{
  digitalWrite(led, HIGH);       // briefly flash the LED
  while(Wire.available() > 1) {  // loop through all but the last
    char c = Wire.read();        // receive byte as a character
    Serial.print(c);             // print the character
  }
  int x = Wire.read();           // receive byte as an integer
  Serial.println(x);             // print the integer
  digitalWrite(led, LOW);
}

finncat62, 2022-03-29:
I added a define above everything for each function(could you explain why I have to do this?)
[image]

It uploaded and worked. YAY!

luni64, 2022-04-01:
Sorry, was quite busy this week. Regarding the required definitions: 

c/c++ needs to see a declaration before it can use an 'item'. You are probably used to do this for variables. I.e., you would expect an error in the following code:
```c++
void someFunc()
{
   i = 42;
   int i;
}
```

Same is valid for functions. The compiler can not know what a symbol like `someFunc` means before you declared it somewhere.  In small, one-file programs you usually sort your functions in the correct way so that you don't have to declare them. 

Works without declaration:
```c++
void funcA()
{
  // do something;
}

void funcB()
{
  funcA();
  // do something else
}

void setup()
{
    funcA();
    funcB();
}
```

needs declarations:

```c++
void funcA();
void funcB();

void setup()
{
    funcA();
    funcB();
}

void funcB()
{
  funcA();
  // do something else
}

void funcA()
{
  // do something;
}
```
In lager projects you would organize your code in various files. In this case you would have the definition  your code in *.cpp files and the declarations in a header *.h.  Code which want to use the functions in your *.cpp file would #include the corresponding header and thus gets the declarations on top. 

The Arduino IDE tries to automate this process behind your back (not a good idea IMHO). So, maybe you didn't stumble over this normal c++ behavior so far. 

Example: 
*Source file: myfunctions.cpp*
```c++
#include "myfunctions.h"

void funcA()
{
  // do something;
}

void funcB()
{
  funcA();
  // do something else
}
```
*Header file myfunctions.h*

```c++
void funcA();
void funcB();
```

*main.cpp*

```c++
#include "myfunctions.h"

void setup()
{
   funcA();
   funcB();
}
```

closed 2022-04-07 by luni64