luni64/VisualTeensy #73: how to edit libraries

finncat62, 2022-07-02:
Hi Luni, it's me again(sorry)

This one should be super simple and fast.
context:
I'm using a library that replaces another (teensy4_i2c instead of wire), but some libraries depend on wire. Luckily, the creator of i2c said that I just have to go to the libraries and replace references to wire to the new library. 

question:
the way that visual studio code starts puts all the teensyduino and arduino libraries in the lib folder, but I don't know where the others are?  Is there a way I can edit them for just inside this project, leaving them the same for all other projects that include those libraries?

[image]

[image]

there is one place that I see reference to wire
[image]

but wire isn't even included in the file?

luni64, 2022-07-03:
What do you mean by "the others"`? According to your vsteensy.json you only use SPI and wire. The screenshot of your lib folder shows EEPROM, SPI and Time.  The screenshot of main.cpp shows that you want to a lot of other libraries. So, the 3 screenshots are kind of completely unrelated? 

Here a video showing how to set up libraries https://github.com/luni64/VisualTeensy/wiki/Using-Libraries. As long as you don't use libraries from the "Shared libraries" repository they will all end up in the lib folder.

finncat62, 2022-07-03:
Yes, I've set up the libraries in visualTeensy in that section.
My question is this: 
where can I find the libraries included?
- The libraries from teensyduino and arduino libraries show up in the lib folder (photo project on the left, lib with eeprom, spi, time, on the right PWMServo, Wire).  However,
- libraries that have been added from the "installed, shared Libraries" section don't appear with the others. In the image on the left side there are libraries such as NewPing, Gfx, etc, which I don't see. Where I can find them? In the image on the right, they are under .vsteensy, build, lib. However, on the image on the left, .vsteensy => build is empty.
[image]
(Image shows two projects side by side, each with libraries from teensyduino and installed. One shows the installed libraries inside the .vsteensy => build  => lib, the other .vsteensy build is empty). 

(started a new project for photos of the process)
libraries added:
[image]

.vsteensy build empty, libraries not visable in main lib
[image]

the goal is to go into the libraries and replace all references to the wire library with references to the replacement library, but I'm not seeing it.

Sidenote: I started both project folders the exact same way, why might they be different?

Thank you very much for the help and paitence,
~finncat62

finncat62, 2022-07-04:
*another thing: I created a new project that only had wire included, not the teensy4_i2c. For some reason, it still shows up in the path, and I get an error saying multiple definitions of wire? What should I do? Wire and teensy4_i2c are exclusive, so having both at once creates issues, but I didn't include it and it's still making problems? 
[image]
[image]

luni64, 2022-07-04:
Only libraries you select from the Teensyduino and the Arduino Repositories are copied to the project folder. Shared Libraries are used from the shared library folder. If you didn't change the configuration they should be under USERDIR\Documents\Arduino\libraries

I suggest to remove **all** the libraries from the shared library folder (shared libraries are a bad idea anyway) 
Then start a new project and include the libraries you need as shown in the video I linked. After saving the project they should be copied into your project/lib folder. As you know, there is no automatic library detection. Some libraries are dependent on other libraries, so you might need to add those libraries as well. Usually the compiler gives some very clear indications on which libraries it doesn't find. Also have a look (F12) into the headers with red wiggles.  It will show you which libraries are still missing. 

BTW Looking in the build folder is not very informative. This only shows the compilation results. The important files are the ones in .vsteensy/,  .vscode/ and the makefile.  If you can't get it compiling please zip the complete project and post here. I can then have a closer look

luni64, 2022-07-04:
> *another thing: I created a new project that only had wire included, not the teensy4_i2c. For some reason, it still shows up in the path, and I get an error saying multiple definitions of wire? What [...]

By " I created a new project" you mean that you started with an empty project folder right?  

If opening a project VisualTeensy checks the /lib folder and automatically includes libraries it found there in the library folder. Also it never deletes code. So, if you start a new project in an existing folder  which already contains libraries you will have them in your project. Other than this I don't see any possibility how a library would sneak into your project.

finncat62, 2022-07-04:
Cool. How do I change the file path here to the project library file: 
[image]

luni64, 2022-07-05:
It reads it out from the Arduino preferences. In the Arduino IDE go to File|Preferences and change the Arduino Sketchbook folder to any convenient location.

finncat62, 2022-07-05:
[image]

[image]

...?

luni64, 2022-07-05:
Do you have a library folder in the new folder?

finncat62, 2022-07-06:
perfect all good:)

to anyone seeing:
In arduino go to file => preferences, change the location to the place you want to have the new libraries. create a file called libraries and include everything inside, then visualTeensy will automatically change to search that when including :)

Thank you, sorry for the trouble.

closed 2022-07-07 by finncat62