luni64/VisualTeensy #74: concatenate multiple .ino files

jdavis-codes, 2022-07-06:
I'm working with a pre-existing project, whose code is structured purposefully to work with the Ardunio IDE. There is a main loop .ino file, and several functions contained in their own .ino files. all are contained in the same src folder.

My understanding is that the arduino IDE concatenates all these files into one in the pre-processing step ([source](https://arduino.github.io/arduino-cli/0.21/sketch-build-process/#pre-processing)).

Is there any way to customize the make file that VisualTeensy creates to allow that same behavior? I've tried changing SKETCH_NAME to the name of the src folder with all .ino files, which did not work.

luni64, 2022-07-07:
Did you try the Arduino Builder mode? 

[image]

This uses the Arduino builder and should give you the same results as with the Arduino IDE.

jdavis-codes, 2022-07-18:
I figured out that if you edit the SKETCH_NAME to equal the direct path to a folder containing all the .ino files (C:\Users\name\Documents\sketch_folder) it will compile fine with arduino builder :) 

In the open and save project function, it might be nice functionality to be able to select a sketch folder and not just auto detect or create a single sketch file. But I could see how that would clash with opening a vanilla c++ project.

closed 2022-07-18 by jdavis-codes