luni64/VisualTeensy #78: Specify port number for each project with Teensy uploader

putian9935, 2023-10-05:
Hi Luni, 

I am finding VisualTeensy a great software and, currently, my computer connects to three Teensy boards (say, A, B, C) running different programs from different directories (hence different makefile). 

Since I can't find a way to specify the port number before uploading with teensy.exe, each time I want to edit the firmware of one board (say, A), I disconnect the other two (B, C) and uploads. I also try to leave all connected and the program begin to guesses which port to use. Sometimes the guess is correct but sometime it is not. 

May I know if there is anyway to edit the makefile such that the port info could be included? I know tycmd can do it but I am just wondering if the same can be achieved with teensy.exe. 

Maybe this is not the right place for this question but any help is much appreciated. Thank you.

luni64, 2023-10-05:
Better to use TyTools to upload. They handle multiple teensies without hassle https://github.com/Koromix/tytools

putian9935, 2023-10-09:
I see. Thank you very much!

closed 2023-10-09 by putian9935